IFL-CAMP/iiwa_stack #229: configuration service not working

gioppitt, 2019-09-25:
Hi everyone, 
I am  struggling with making configuration servers work. I've tried some of them and no one seem to influence the way my robot works. In particular, I would like to limit the speed, for safety reasons.

Here is an example of how I am using the service:

```string client_str = "/iiwa/configuration/setPTPCartesianLimits";
    ros::ServiceClient client = robot_node->serviceClient<iiwa_msgs::SetPTPCartesianSpeedLimits>(client_str);
    iiwa_msgs::SetPTPCartesianSpeedLimits config;
    config.request.maxCartesianVelocity = speed;
    config.request.maxOrientationVelocity = speed/10;
    config.request.maxCartesianAcceleration = -1;
    config.request.maxOrientationAcceleration = -1;
    config.request.maxCartesianJerk = -1;
    config.request.maxOrientationJerk = -1;

    if (client.call(config)) {
        if(!config.response.success)
            ROS_ERROR_STREAM("Config failed, Java error: " << config.response.error);
        else
            ROS_INFO_STREAM("Path Parameter Service successfully called.");
    }
    else {
        ROS_ERROR_STREAM("Config failed - service could not be called - QUITTING NOW !");
    }
```

Everything "seems" to be working correctly, as I get the output "Path Parameter Service successfully called." on my terminal. However, even if "speed" is set to 0, the robot moves.

Thanks in advance for your help.

Giovanni

exo-core, 2019-09-30:
Hi Giovanni,

the `SetPTPCartesianSpeedLimits` only influences the speed for point-to-point motions that are used for the ROS action interface. If you control the arm via MoveIt! or and other ROS based controller that sends position targets with a certain frequency you need to change the SmartServo speed limits instead using the `SetSmartServoLinSpeedLimits` service.

gioppitt, 2019-09-30:
I see your point, but I had already tried every service and no one acts on the speed.

However, I tried again what you said and no effect: I succeed in calling the service but there is not effect on the robot's motion. Any other suggestion?

Thank you very much.

havefun28, 2019-11-27:
I have the same problem. Calling the service in the command line instead of in scripts works for me. ANy help with calling the service in scripts will be greatly appreciated.

CodingCatMountain, 2022-09-01:
@RuihanGao  Hi, Have you try to change to JointImpedanceControl mode via  Calling the service in the command line?? Each time, I do it return a fail...

RuihanGao, 2022-09-01:
> 

Sorry I haven't tried it yet. I only tried to use the Impedance control in manual mode.