IFL-CAMP/iiwa_stack #231: how to control lbr 14 with torque/effort control model

921218316, 2019-10-12:
I would like to control iiwa lbr 14 with torque/effort control model,that is to say is there any functions that take every joint' torque/effort as input to control it in the iiwa stack package？Thank you very much！

jacknlliu, 2019-12-10:
@921218316 you can consider this repo [epfl-lasa/iiwa_ros](https://github.com/epfl-lasa/iiwa_ros)

nikhilpodila, 2022-01-27:
@jacknlliu this surprisingly works. 

I noticed that the URDF in the IFL-CLAMP/iiwa_stack repo has a few features different from epfl-lasa/iiwa_ros. Most noticeable ones being:

- Max effort of all joints are too small (not enough even for gravity compensation).
- `safety_controller_k_vel` parameter is small.
- Collision models are complex (I found [an external resource](https://adamconkey.medium.com/collision-meshes-for-simulating-robots-in-gazebo-ccc647d8b87d) that gives some explanation on this).

Is there a reason why these two repositories differ so much in terms of the robot model?