IFL-CAMP/iiwa_stack #233: CartesianPose not working

lin2jy, 2019-10-25:
Hi, the following cpp file worked fine for calling and getting Cartesian positions before, but after we installed and run ROS in another PC than the old one (having Sunrise installed in Windows) we can't get Cartesian positions anymore. Our Sunrise and ROS are installed in two different PCs and we found a way to clone everything. Now, I can obtain Cartesian poses directly on the terminal using "rostopic echo /iiwa/state/CartesianPose", but I can not obtain any Cartesian pose message in the following cpp file therefore causing inability to control KUKA by Cartesian control. Again, this same cpp file worked fine without any problem before. I need this Cartesian control working in cpp file so I can do further more complicated work with the KUKA arm. Any help is appreciated. Thanks in advance!
----------------------------------------------------------------------------
#include "ros/ros.h"
#include "std_msgs/String.h"
#include "geometry_msgs/PoseStamped.h"

#include <iiwa_ros.h>
#include <conversions.h>
 
geometry_msgs::PoseStamped my_cartesian_pose;

iiwa_ros::iiwaRos pheebs_iiwa;

void getcart(const geometry_msgs::PoseStamped::ConstPtr& value)
{       
  my_cartesian_pose = *value;
  ROS_INFO_STREAM("I got a Cartesian Pose and its contents is :" << my_cartesian_pose.pose);	
}

int main (int argc,char **argv)
{
  ros::init(argc,argv,"iiwa_ros"); // Initialize ROS

  ros::NodeHandle n;

  ROS_INFO("Started");	

  pheebs_iiwa.init(); // Initialize iiwa_ros class

  ros::Subscriber sub_cart = n.subscribe("iiwa/state/CartesianPose", 1, getcart);

  ros::Rate loop_rate(0.5);

  int count = 0;
  while (ros::ok())
  {	
    // Creat home Cartesian pose	
    my_cartesian_pose.pose.position.x = 0;
    my_cartesian_pose.pose.position.y = 0;
    my_cartesian_pose.pose.position.z = 1.331999998;
    my_cartesian_pose.pose.orientation.x = 0;
    my_cartesian_pose.pose.orientation.y = 0;
    my_cartesian_pose.pose.orientation.z = 0;
    my_cartesian_pose.pose.orientation.w = 1.0;

    if(count > 5)
    {
      // Create Cartesian pose
      my_cartesian_pose.pose.position.x = 0.55;
      my_cartesian_pose.pose.position.y = 0.04;
      my_cartesian_pose.pose.position.z = 0.09;
      my_cartesian_pose.pose.orientation.x = -0.19;
      my_cartesian_pose.pose.orientation.y = 0.97;
      my_cartesian_pose.pose.orientation.z = -0.04;
      my_cartesian_pose.pose.orientation.w = -0.09;
    }	    		     

    ROS_INFO("publish, %d", count);

    pheebs_iiwa.setCartesianPose(my_cartesian_pose); // Set Cartesian pose

    ros::spinOnce();

    loop_rate.sleep();	
    count++;	    
  }

   return 0;
}

irgusite, 2020-07-22:
Hey! 
Same problem here, I can't get the node to get data from the robot, but I see if I do a rostopic echo /iiwa/state/CartesianPose. 
Did you solve the problem? I also see that my node is subscribed to the topic when it runs...

leon-no37, 2021-04-07:
Please use the correct type, which is "iiwa/CartesianPose".