IFL-CAMP/iiwa_stack #235: How to achive cartesian velocity control?

tasbolat1, 2019-11-23:
Hello community,
I am having trouble to  setup cartesian velocity control for KUKA IIWA 14. My task is to my robot's end effector linearly between two cartesian points with constant speed. So far, I could not find any solution. I have seen CartesianVelocity msg file, but no command for CartesianVelocity exists and I am not sure what hardware interface shall be used. Any suggestions and recommendationsa are welcome. Thanks.
Kind Regads,
Tas

emielke12, 2020-01-08:
I have not tested it, but there is a feature/CartesianVelocity branch you could try out.

Alternatively, I have had success using PyKDL and the iiwa14 urdf files to create a jacobian matrix based on the current joint position. Then using pseudo-inverse inverting the jacobian and multiplying the desired cartesian velocity to get the joint velocities needed, and then publishing the joint velocities to /iiwa/command/JointVelocity.

BE ADVISED:  you need to be very careful commanding the robot in this way, as it can easily collide with surroundings if you do not know what you are doing. I would only advise an advanced user to try to implement something like this.

eratner, 2021-08-28:
Hi @emielke12, I am also trying to achieve Cartesian velocity control for the Kuka iiwa 14. I am following up because I have tried a method similar to what you suggested-- using a Jacobian pseudoinverse to map from desired Cartesian velocity to joint velocities, and then publishing the resulting joint velocities to the `/iiwa/command/JointVelocity` topic. The problem is that I am finding the motion to be very jerky (similar to what is described in #128). In my case, my desired Cartesian velocities are very low, and I have seen the same jerky behavior for various publishing frequencies. Would you mind sharing more details/code on how you were able to achieve good performance using this method?

Thank you so much!

emielke12, 2021-08-30:
@FaridTavakol 

> One thing to mention is that I can directly send a CartesianVelocity command using rostopic pub command in the terminal, and the robot seems to have no issues with the message. But it throws the error when I run my node.

I'd say your node would be the first place to look then. It looks like its not getting serialized correctly, if I am reading those errors right. Do you change anything in the message between the rostopic implementation and the node? Hard to say without seeing the code, but I'd double check to make sure you've initialized the node properly, and have the same message type as the rostopic implementation.

FaridTavakol, 2021-08-30:
@emielke12 Thanks for your reply. Yeah, that issue is resolved after I switched back to the `TwistStamped` type. The velocity control motion in general is pretty laggy and not smooth like the position commands. However, when I use the terminal to publish the command for the Cartesian velocity the motion seems pretty smooth! I played around with my ros rate and queue size but it didn't make big difference.

emielke12, 2021-08-30:
@eratner I don't recall having any jerky behavior, but it could be the latency of communication. This iiwa driver does not use the FRI (Fast Robot Interface) which gives access to the low-level real-time control. So I would check to see if your communication is actually getting the messages over in time. 

My implementation was using PyKDL to create a kinematic chain and to calculate the jacobian, then using NumPy pseudoinverse to calculate desired joint velocities. Node was publishing at 100 Hz, but I haven't used this in a while...

eratner, 2021-09-01:
@emielke12 Thank you very much for the reply. Do you have any recommended resources/information on how to use the FRI for these purposes?

emielke12, 2021-09-01:
@eratner Lately I've been using [this](https://github.com/epfl-lasa/iiwa_ros) to get access to torque control, give it a look (it uses the FRI)