IFL-CAMP/iiwa_stack #239: Pre-program kuka iiwa14 trajectory trough Gazebo

MaMarcelloChiurazzi, 2020-01-27:
Hello to everyone and thanks for all the amazing work that has been done on this package.

I have got one question regarding the trajectory planning of the Kuka.

My idea is to set up a static environment in which the robot has to operate; On this regards I am building, using gazebo, a virtual environment merging the urdf.xacro of the robot (whitin the iiwa_description folder) together with other custom models builded up as urdf or sdf.
Clearly, each object within the environment has its own collision description.

My idea is to use the bidirectional communication between ros and Kuka cabinet for:

1) Monitor and update the real robot configuration from Kuka cabinet to Gazebo;
2) Send using Ros the goal_pose the robot has to reach;
3)Use the ik developed by kuka for simulating in Gazebo the trajectory that allows the robot to reach that demanded pose without colliding with other objects within the environment;
4) When (and if ) found one, send the trajectory to the real robot to perform the motion.

For what I have understood, step 1,2 and 4 are already implemented. I am wondering to know if there is something like step 3 already implemented.

Thanks in advance for your precious help.

Regards

exo-core, 2020-01-27:
Hi @MaMarcelloChiurazzi,

have a look at the `iiwa_stack` packages for MoveIt! integration. They offer the functionality you are looking for.

MaMarcelloChiurazzi, 2020-01-27:
Thanks for the reply @exo-core 
Does move-it integrates the low-level kuka inverse kinematic? 
I have used Move-it in the past with a PR2 robot and an UR5 but sometimes the trajectory performed were really weird (at the end you were able to reach the demanded pose but performing an huge movement).

JimmyDaSilva, 2020-01-27:
> I have used Move-it in the past with a PR2 robot and an UR5 but sometimes the trajectory performed were really weird (at the end you were able to reach the demanded pose but performing an huge movement).

@MaMarcelloChiurazzi, I experience that problem in the past. This is happens when the IK solver gives an answer which is not the closed one to the current articular positions.
So basically this depends on which IK solver you use inside MoveIt, and not direcly MoveIt.

For example, trac-ik has a Distance solve type, which gives you the closest solution to the current one.
Tutorial to use it with MoveIt:
http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/trac_ik/trac_ik_tutorial.html

> Does move-it integrates the low-level kuka inverse kinematic?

I don't know but my guess is that the code for the low-level IK from KUKA is not accessible. So probably not usable at runtime even if you manage to call it.

MaMarcelloChiurazzi, 2020-01-28:
Perfect, I have never tried this ik solver, I will give a try to it.
Thanks very much for your reply @JimmyDaSilva