IFL-CAMP/iiwa_stack #241: Compatibility Question for ROS Melodic Morenia on Ubuntu 18.04.4 LTS (Bionic Beaver)

luke-kim409, 2020-03-17:
Hi there! 

I am trying to use this iiwa_stack for my new LBR. I am using ROS Melodic Morenia on Ubuntu 18.04.4 LTS (Bionic Beaver). On the Wiki instruction (setup guide), this stack seems to be compatible with ROS Melodic.

I was able to get the robot and ROS machine (my laptop with Roscore) connected; I check the log message saying that the connection was made on the SmartPAD when running both roscore and SmartServo at the same time. 

However, I was not able to read the robot state through commands (rostopic echo /iiwa/state/JointPosition) on the terminal, because I couldn't set up the ROS Java messages detailed in [addmessage](https://github.com/IFL-CAMP/iiwa_stack/wiki/addmessage) wiki page.  I was not able to successfully build the workspace in my environment, and therefore I was not able to find the created jar file in the directory (src/iiwa_msgs/build/libs/), which even didn't exist probably due to the unsuccessful catkin build.

I think [ROS Java](http://wiki.ros.org/rosjava) tool currently does not supprot ROS Melodic, but I see some folks are using ROS Melodic with this stack.

Can anyone who has experience of using ROS Melodic with this stack give some tips/advice to test simple tasks such as read robot state and command joint motions on the terminal (to test whether my ROS+LBR system works)? 

Many thanks in advance for the help!

luke-kim409, 2020-03-18:
I tried with ROS Kinetic on Ubuntu 16.04.6 LTS (Xenial Xerus) as well, but got stuck at the same point (highlighted part in the snapshot below) in the [addmessage](https://github.com/IFL-CAMP/iiwa_stack/wiki/addmessage) wiki page. 
[image]

The catkin build was not successful (or partially successful) shown below:
[image]

After this partially successful catkin build, I was able to find 'iiwa_msgs' directory under 'src'.
However, unlike what is described in the wiki page, I found build.gradle file only (no further sub-directories and jar file).

I believe, with this, I may not be able to read the robot state through commands (e.g. rostopic echo /iiwa/state/JointPosition) on the terminal. 

Can anyone help me to figure out how to resolve this issue? Thanks!

exo-core, 2020-03-18:
Hi @luke-kim409,

ROSJava and melodic are indeed no best friends. What worked for me, is to build the jar-files using kinetic, as they are fully compatible.

However, there is actually no need to build them yourself at all, as there are binaries already available under in `iiwa_ros_java/ROSJavaLib`.

luke-kim409, 2020-03-30:
Hi @exo-core, 

Thank you for your response. 
It turned out that I did not source the workspace appropriately before typing the commands on the terminal. Now I am able to control my arm with my Ubuntu 16.04 machine with ROS Kinetic. As you mentioned, iiwa_stack works without building ROS Java separately as long as one copies ROSJavaLib into the Sunrise project. Thanks again!

I have another Ubuntu 18.04 machine with ROS Melodic, and it seems from #141 that after the master branch was merged into Melodic development branch, the current master branch as it is should work with Melodic. When I tried to connect the same robot arm with my Ubuntu 18.04 machine with ROS Melodic, however, I was not able to get the robot connected. The robot was just waiting for the ROS Master to be connected even when I ran the roscore. I believe I am now quite familiar with the setup, and I checked that network configuration was fine. Everything was the same as what I did for my Ubuntu 16.04 with ROS Kinetic. 

I have no clue why I cannot connect my robot with ROS Melodic.
Could you share some insights?

Thanks!

JimmyDaSilva, 2020-03-30:
> I have no clue why I cannot connect my robot with ROS Melodic even though I followed the exact same step as I did for ROS Kinetic (with which I was able to get my robot connected).
> Could you share some insights?

For starters, I would try to ping the kuka cabinet from the Ubuntu computer to check if there is no communication issue.

luke-kim409, 2020-03-30:
Hi @JimmyDaSilva,
I checked that the network configuration was correctly done. I was able to ping the KUKA sunrise cabinet from my Ubuntu desktop with ROS Melodic, but the robot kept waiting for the ROS master to be connected. When I switched to my Ubuntu laptop with ROS Kinetic, the connection was made pretty well. I am wondering what the problem would be.

JimmyDaSilva, 2020-03-30:
Now that I am thinking of it, I might have had the same issue.
Have you tried not using the same IP address for your laptop and your desktop?

i.e, if it works for your laptop with `172.31.1.100`, use `172.31.1.101` with your desktop computer.
Remember you'll have to change the IP in `src/RoboticsAPI.data.xml` or by using the SmartPad in the Process Data window.

If this does not work, you could try to add the robot IP in the hosts file.
In `/etc/hosts` add the line:
```172.31.1.147    iiwa```

Good luck !

luke-kim409, 2020-03-30:
Thank you for your suggestion @JimmyDaSilva!
Actually, I tried using different IP addresses for my laptop (172.31.1.148) and desktop (172.31.1.150) while changing the IP in the Process Data from the SmartPAD accordingly and trying ping before running the roscore. It didn't work for me unfortunately. I change the IP address of my laptop with ROS Kinetic to 172.31.1.150 and then the robot could be connected again.

I just tried to add the robot IP in the hosts file, but sadly it seems nothing has changed.

luke-kim409, 2020-04-20:
I still cannot connect my robot on ROS Melodic. 
I will close this so that I can start from a simpler question.

closed 2020-04-20 by luke-kim409