IFL-CAMP/iiwa_stack #243: Dart physics engine incompatibility

arturShaf, 2020-03-27:
Hi everyone!
I'm trying to launch simulation via `roslaunch iiwa_moveit moveit_planning_execution.launch` with Dart physics engine however after I set target pose via RViz and try to _Plan and Execute_ it, the model executes most of the motions and right when it reaches target pose instead of stopping it starts to spin chaotically and then collapses (see the attached picture). At this point for each link (i.e. 1 to 7) the following errors start to pop up:
`Ignoring transform for child_frame_id "iiwa_link_1" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan -nan -nan -nan)`
[image]
I probably should mention the following:

- When using ODE physics engine everything works perfectly well.
- There are two errors when launching `moveit_planning_execution`:
     1. `No p gain specified for pid.  Namespace: /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_1` -                  this one appears when I use ODE physics engine as well (and it still works). I tried [this](https://github.com/IFL-CAMP/iiwa_stack/issues/161) and [this](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/886) solutions but they didn't work for me. But as far as I understood this warning means that some default Gazebo behavior will be used. 
     2. `ROS get_physics_properties service call does not yet support physics engine [dart].`
- Once in a while KUKA manages to complete first trajectory execution and let me choose the next state, but then collapses anyway.

I use Ubuntu 18.04 with ROS Melodic and Gazebo-9.

Any help and feedback would be appreciated, thanks in advance!

closed 2020-04-08 by arturShafikov

huangdii, 2020-11-19:
how to solve the problem? I have same issue :(

arturShafikov, 2020-11-21:
> how to solve the problem? I have same issue :(

Hello, Huangdii!
It's been a while so I might be wrong, but as far as I remember, changing _joint_damping_ value in _iiwa_description/urdf/iiwa14.xacro_ or in _iiwa_description/urdf/iiwa7.xacro_  to 10.0 for example (instead of 0.5) solves the problem. I think it's not correct for this value to be lower than 1.0. 
If it's not the case, feel free to write againg, I'll try to think of something else.