IFL-CAMP/iiwa_stack #244: Cannot implement Spline motion in Cartesian impedance control mode

LeeKeyu, 2020-04-10:
Hi community,
I am recently trying to execute spline motion under the Cartesian impedance control mode with iiwa7.
I wrote a python script to achieve the same function as this example.
[spline motion demo](https://github.com/exo-core/iiwa_stack_examples/blob/development/iiwa_actions/src/SplineMotionDemo.cpp). When no control mode is specified (by default use the Position Control mode), it worked.
However, when I configured the control mode to be Cartesian impedance control before executing the spline motion, iiwa simply doesn't move, no error was reported on my PC or SmartPad, and I receive this traceback after exiting:
Traceback (most recent call last):
  File "/home/likeyu/RoboticUS/roboticUS_ws/src/kuka_control/kuka_ros/src/test_motion_action.py", line 222, in <module>
    if not splineMotionClient.get_result().success:
AttributeError: 'NoneType' object has no attribute 'success'

Can anybody offer some help? Thanks in advance.

Here is my cartesian control mode parameters:
        iiwa_config_srv_2 = ConfigureControlModeRequest()
        iiwa_config_srv_2.control_mode = 2 # cartesian impedance
        iiwa_config_srv_2.cartesian_impedance.cartesian_stiffness.x = 5000
        iiwa_config_srv_2.cartesian_impedance.cartesian_stiffness.y = 5000
        iiwa_config_srv_2.cartesian_impedance.cartesian_stiffness.z = 5000
        iiwa_config_srv_2.cartesian_impedance.cartesian_stiffness.a = 300
        iiwa_config_srv_2.cartesian_impedance.cartesian_stiffness.b = 300
        iiwa_config_srv_2.cartesian_impedance.cartesian_stiffness.c = 300
        iiwa_config_srv_2.cartesian_impedance.cartesian_damping.x = 1.0
        iiwa_config_srv_2.cartesian_impedance.cartesian_damping.y = 1.0
        iiwa_config_srv_2.cartesian_impedance.cartesian_damping.z = 1.0
        iiwa_config_srv_2.cartesian_impedance.cartesian_damping.a = 1.0
        iiwa_config_srv_2.cartesian_impedance.cartesian_damping.b = 1.0
        iiwa_config_srv_2.cartesian_impedance.cartesian_damping.c = 1.0
        iiwa_config_srv_2.cartesian_impedance.nullspace_stiffness = 100
        iiwa_config_srv_2.cartesian_impedance.nullspace_damping = 1.0

LeeKeyu, 2020-04-11:
Still haven't figured it out. 
I changed the control mode by calling the service /iiwa/configuration/ConfigureControlMode in the python script before executing the action. The control mode is successfully changed, but it cannot execute the spline motion action. As long as I comment the code that calls this service, the action can be completed successfully.
I also tried to use the terminal to call the service ConfigureControlMode before running the node, but changed nothing.
Can anybody help? Many many thanks!

LeeKeyu, 2020-04-13:
I seem to understand a little bit about this problem. It should be caused by different command types. The actions are using the PTP commandtypes (e.g., POINT_TO_POINT_CARTESIAN_POSE, POINT_TO_POINT_CARTESIAN_SPLINE), while after configuring control type, the motions that can be executed seem to be only the smartservo command types such as SMART_SERVO_CARTESIAN_POSE_LIN.
I tried to change the type of MoveToCartesianPoseActionGoal to be SMART_SERVO_CARTESIAN_POSE_LIN in the iiwaActionServer.java, and now I am not getting the traceback "AttributeError: 'NoneType' object has no attribute 'success'"  anymore, but unfortunately the action still cannot be executed. I think I will just give up using action to send cartesian pose commands, and only use the topic.
If anyone has figured out how to use the action goals in Cartesian impedance control mode, I would really appreciate your help and suggestions. Thanks a lot!

michelepenzo, 2020-12-09:
HI @LeeKeyu , same problem. Solved?

LeeKeyu, 2020-12-09:
> HI @LeeKeyu , same problem. Solved?

Not yet.

michelepenzo, 2020-12-09:
are you still working on the problem?

LeeKeyu, 2020-12-10:
> are you still working on the problem?
> 
> I'm having the same issue, when I switch from position control to joint impedance control the first time works, but the second one actionClient stuck on waiting result..

Hi, I'm not working on it any more, and I was only using the topic instead of action and client to send Cartesian commands under the impedence control mode. Hope the maintainers can fix it someday.

RuihanGao, 2021-05-22:
> are you still working on the problem?
> 
> I'm having the same issue, when I switch from position control to joint impedance control the first time works, but the second one actionClient stuck on waiting result..

Hi, I am having a similar issue that swithcing between action client and smart_servo command (topic) does not work after the first time. Did you solve it or have any suggestions? I try to use action to send Goal of tool Frame and get response when the motion is completed. Thanks in advance!

CodingCatMountain, 2022-09-01:
@LeeKeyu Hi, May I ask a question about change the control mode ? I use this command ``rosservice call /iiwa/configuration/ConfigureControlMode`` in the shall, each time it just return fail. How can I solve this problem??