IFL-CAMP/iiwa_stack #246: My robot gets connected in ROS Kinetic, but NOT in ROS Melodic, why?

luke-kim409, 2020-04-21:
Hello,

I am having trouble in getting my robot connected with ROS Melodic. I have a laptop with ROS Kinetic on Ubuntu 16.04, and the robot works perfectly well. For my desktop with ROS Melodic on Ubuntu 18.04.4, however, the robot keeps waiting for the ROS master to be connected even when I run the roscore. 

In my previous issue #241, I got some advice from @JimmyDaSilva on things to check or try.
Let me summarize what I tried and checked so far:

1) Network ping was successful -- I was able to ping the KUKA Sunrise Cabinet (172.31.1.147) from my desktop, but couldn't make the connection.

2) I used the same network configuration for my laptop and desktop (both with IP address 172.31.1.148). The robot was connected to the laptop with ROS Kinetic, but failed to get connected to the desktop with ROS Melodic. 

3) I tried using another IP address for my desktop (172.31.1.150) with ROS Melodic, while changing the IP in src/RoboticsAPI.data.xml accordingly (or in the Process Data window on the smartPAD), but it didn't make any change (the robot still waited for the ROS master to be connected). When I try this new IP address for my laptop with ROS Kinetic, the connection was made quickly as soon as I ran the ROSSmartServo application on the smartPAD.

4) After this, I also tried to add the robot IP (172.31.1.147) in the hosts file by adding 
a line "172.31.1.147 iiwa" in /etc/hosts, but it didn't work either.

5) I tried completely removing ROS Melodic and reinstalling it, but the connection problem persists.
I haven't tried reinstalling Ubuntu 18.04.4 to start from the very beginning.

Can anyone confirm whether the iiwa_stack works with ROS Melodic as it is without any specific things to do additionally? Has anybody experienced this connection problem? 
Any advice or comment would be greatly helpful for me! Many thanks in advance!

TayaIgluna, 2020-06-30:
Hi ! By any chance, did you find a solution to this problem ? We have several computers running on ROS melodic and we want to connect our iiwa to it.

luke-kim409, 2020-07-04:
Unfortunately not yet. Having no idea how to get it connected to ROS Melodic, I am just running the robot on ROS Kinetic instead.

TayaIgluna, 2020-07-04:
We had the rosmaster connected and we were able to receive topics but not send commands. We now made it work but are not sure why it's working now.
For your connection to rosmaster, the problem is almost surely your IP addresses. 
The computer running on melodic has 172.31.1.150 as IP. The iiwa is on default IP address. We created a new sunrise project with nothing inside except iiwa_stack. It wasn't working and we tried again the day after and it was working. So my advice would be, first check all your IP, start a new sunrise project from scratch, source everything in every terminal you open and restart computer & robot a few times in between. 
Good luck and kind regards !

irgusite, 2020-07-04:
And other point we discovered working on it, is that if the node crashes on the cabinet, you have to reboot the whole system to get it working again. Even if it doesn't clearly show the crashing (like not being able to connect to the ROS master. 

Hope you get it working :)

luke-kim409, 2020-07-07:
Thank you both! I will try with a new sunrise project from scratch to see if it works. I will update if I can make the connection.

luke-kim409, 2021-03-11:
@TayaIgluna @irgusite @dornbirndevelops Hello all, I tried to follow your suggestions/discussions, but I wasn't able to get my robot connected to ROS Melodic. I created a new Sunrise Project from scratch and removed seemingly unnecessary items in the Sunrise project except for the required ones in [iiwa_ros_java](https://github.com/IFL-CAMP/iiwa_stack/tree/master/iiwa_ros_java). Network setting was corretcly done; I was able to ping the KUKA controller from my Ubuntu desktop with ROS Melodic. But still I cannot get it connected to ROS Melodic when I run **rosecore** and the **ROSSmartServ**o application on the smartPAD. When I change to my laptop with ROS Kinetic, it just runs without any problem. I have no idea what causes the problem. Could you please give any further advice based on your experiences so far? Are your machines working well without any issues?

matthias-mayr, 2021-03-15:
One thing that you could double-check is that you are using IP addresses to identify ROS machines at every point of your setup. So make sure that `ROS_HOSTNAME` is not set and only `ROS_IP` is used for both the cabinet and the computer.

I tried to use hostnames first, because it seemed to be a cleaner solution. But then I ran into issue with `iiwa_stack`. iirc, there were first connection issues and then also a state in which I could get the status, but not command the robot.

HYEDO-KIM, 2021-03-29:
@luke-kim409  Hi!! I'm trying to jogging arm in Real-Time in ROS melodic.
But I'm having a hard time right now.
I think with your help, things will be easier.
Can you help me?
plz answer T.T
hyesine1107@gmail.com

luke-kim409, 2022-05-12:
It seems the desktop computer I used had some issue with the robot connection.
I tested with another laptop and can confirm that iiwa_stack is compatible with both ROS Melodic and ROS Noetic.
Closing this issue.

closed 2022-05-12 by luke-kim409

Mechazo11, 2022-10-28:
@luke-kim409 Could you move the arm in AUT mode using rostopic pub commands sent out manually from a termial?