IFL-CAMP/iiwa_stack #249: no communication able between KUKA LBR iiwa7 and ROS Melodic

dornbirndevelops, 2020-06-22:
Hi all,

I am experiencing communication issues between my Notebook running Ubuntu 18.04.4 (IP 192.168.20.51 /24) with ROS Melodic, Sunrise Workbench 1.15 (in VM) and my iiwa7 connected directly at X66 port (IP 192.168.20.55).
I've followed the setup guide from the wiki and managed to sync a working ROSSmartServo Application to the Cabinet. The Robot apps xml config has the same IP set for Ros Master as my Notebook. export variables are set as mentioned in the wiki, catkin workspace builds without errors, simulation is working fine, ...

When starting the SmartServoApp, the Robot clicks right after connecting to the roscore. The Info on screen also indicates an established connection. `rostopic list` also shows all the topics the robot has to offer.

But as soon as I listen for any topic via `rostopic echo /iiwa/state/JointPosition` I receive nothing (not even an error).
Running the arms moveit demo `roslaunch iiwa_moveit moveit_planning_execution.launch sim:=false` only results in the following output (last message repeats about every 10 secs)

[image]

Is there something I might have overlooked? I'm out of ideas, any help would be greatly appreciated.

TayaIgluna, 2020-07-04:
Hey ! We managed to make it work, you can check my answer in the similar issue https://github.com/IFL-CAMP/iiwa_stack/issues/246
Good luck !

dornbirndevelops, 2020-07-04:
Thank you for the hint @TayaIgluna I will try it as soon as I possible and let you know if it worked.

dornbirndevelops, 2020-07-14:
@TayaIgluna It works like a charm now. I've followed your advice and created a new Project with only the iiwa_stack sources in it. I've set up the safety configuration, added the software modules needed, installed it and done.
Also it is worth noting that I first used the most advanced fork of this repo (commit-wise) from @exo-core . Now I'm using the base repo as a source. I will try the more advanced fork later on, to see if something is broken.

closed 2020-07-14 by dornbirndevelops