IFL-CAMP/iiwa_stack #251: when using moveit in a simulation environment, an error appears that : Group 'manipulator' was not found.

Mr-rain-drop, 2020-07-16:
Thanks for looking at my problem, I want to command the robot using moveit in a simulation environment, so I ran the following code to start a simulation in gazebo:

**roslaunch iiwa_moveit moveit_planning_execution.launch**

Then, I used the cpp file from the wiki pages of commandrobotmoveit：

**#include "ros/ros.h"
#include <iiwa_ros.h>
#include <moveit/move_group_interface/move_group_interface.h>

using moveit::planning_interface::MoveItErrorCode;

int main (int argc, char **argv) {
  
  // Initialize ROS
  ros::init(argc, argv, "CommandRobotMoveit");
 ros::NodeHandle nh("~");
  
  // ROS spinner.
 ros::AsyncSpinner spinner(1);
 spinner.start();
  
 iiwa_ros::iiwaRos my_iiwa;
 my_iiwa.init();
  
 std::string movegroup_name, ee_link;
 geometry_msgs::PoseStamped command_cartesian_position;
  
  // Dynamic parameters. Last arg is the default value. You can assign these from a launch file.
 nh.param<std::string>("move_group", movegroup_name, "manipulator");
 nh.param<std::string>("ee_link", ee_link, "tool_link_ee");
  
  // Dynamic parameter to choose the rate at wich this node should run
  double ros_rate;
 nh.param("ros_rate", ros_rate, 0.1); // 0.1 Hz = 10 seconds
 ros::Rate* loop_rate_ = new ros::Rate(ros_rate);
    
  int direction = 1;
  
  // Create MoveGroup
 moveit::planning_interface::MoveGroupInterface group(movegroup_name);
 moveit::planning_interface::MoveGroupInterface::Plan myplan;
      
  // Configure planner 
 group.setPlanningTime(0.5);
 group.setPlannerId("RRTConnectkConfigDefault");
 group.setEndEffectorLink(ee_link);
 MoveItErrorCode success_plan = MoveItErrorCode::FAILURE, motion_done = MoveItErrorCode::FAILURE;
  while (ros::ok()) {
    if (my_iiwa.getRobotIsConnected()) {
      
 command_cartesian_position = group.getCurrentPose(ee_link); 
 command_cartesian_position.pose.position.z -= direction * 0.10;
  
 group.setStartStateToCurrentState();
 group.setPoseTarget(command_cartesian_position, ee_link);
 success_plan = group.plan(myplan);
      if (success_plan == MoveItErrorCode::SUCCESS) {
 motion_done = group.execute(myplan);
      }
      if (motion_done == MoveItErrorCode::SUCCESS) {
 direction *= -1; // In the next iteration the motion will be on the opposite direction
 loop_rate_->sleep(); // Sleep for some millisecond. The while loop will run every 10 seconds in this example.
      }
    }
    else {
      ROS_WARN_STREAM("Robot is not connected...");
      ros::Duration(5.0).sleep(); // 5 seconds
    }
  }  
};** 

The code compiled successfully,but failed to run，
**[ERROR] Robot semantic description not found, Did you forget to define or remap '/robot_description_semantic'? 
[FATAL] Group 'manipulator' was not found.**
and I found that there existed /iiwa/robot_description_semantic,but not /robot_description_semantic in the rosparam list,I know that the move group is under the namespace iiwa ,but I really don't know how to fix this problem, I will be grateful if someone can give me a solution.

closed 2020-07-21 by Mr-rain-drop

lyh458, 2020-10-20:
Hi, I have a similar problem here, how did you solve your problem? Thanks.

mf093087, 2022-01-18:
The same to me