IFL-CAMP/iiwa_stack #252: Hand Guide Motion

liushuya7, 2020-07-23:
Is there support for hand guide motion with iiwa_stack? For instance, ROS can observe the robot states as the user manually move the robot joints. I know KUKA has an application to do this task which requires a button to trigger the motion. However, I am hoping to keep ROSServer application running so that it maintain ROS connection.

matthias-mayr, 2020-08-11:
You can use iiwa_stack to configure e.g. a Cartesian impedance stiffness of 0 in all directions. That makes the arm gravity-compensated and free floating. So you can move it manually and read the state of the joints.

CodingCatMountain, 2022-08-31:
@matthias-mayr Hello, sir. May I ask you for more details? Because I have try to run ``rosservice call /iiwa/configuration/ConfigureControlMode`` and modify the ``control_mode`` to 1 , which is the jointImpedanceControl mode. However I only get an error. In this command , I also set the stiffness is 0 and the damping is 0.7. My ROS Version is Noetic, And my robot is iiwa 7 R800.   

 ``

matthias-mayr, 2022-08-31:
We are not using iiwa_stack anymore, but doesn't this page explain most of it?
https://github.com/IFL-CAMP/iiwa_stack/wiki/configuresmartservo

Apparently the topic is called ` /iiwa/configuration/configureSmartServo`