IFL-CAMP/iiwa_stack #265: Real-time control with kuka -iiwa

Bannigo, 2021-04-16:
Hello everyone,
This is not an issue, rather I have a doubt. I am using ROS node to control kuka lbr iiwa. I would like to know if there is real-time functionality with the robot. Currently, I am providing the robot with target joint positions. I can provide a bunch of joint position together and the robot will travel to each and every one of them in a sequence with a specified velocity. But, I want to apply a real-time motion such that even if the robot is moving to a specific joint position it will stop and start executing the most recent position. which function should be used from the sunrise workbench. It would be really nice to know your suggestions. Thank you in advance!!

JimmyDaSilva, 2021-04-16:
Hi @Bannigo,
This iiwa_stack package uses the SmartServo interface of KUKA, which is not real-time.

To do hard real-time control, you need to use the FRI interface. The [iiwa_ros](https://github.com/epfl-lasa/iiwa_ros) package uses this interface. 

Though, even this package suffers from some issues (see [here](https://github.com/epfl-lasa/iiwa_ros/issues/36)) to properly do realtime control. So, in the end, I am currently using my own code without ROS.
We intend to release realtime capabilities this year though with OROCOS+ROS drivers for the IIWA.

Bannigo, 2021-04-16:
Hi @JimmyDaSilva . 
Thank you for your quick reply!! I will try this package out. Is it okay, if I reach you out in the future? I recently got access to Kuka and I am still quite new to this. Seeing the conversation in the attached issue, you seem to be quite familiar with the robot.

JimmyDaSilva, 2021-04-18:
Sure. But don't expect too much, it is the last year of my PhD, I'm gonna be extra busy.
Please, don't forget to close this issue :)

closed 2021-04-23 by Bannigo