IFL-CAMP/iiwa_stack #266: I can read the states but can't command the robot

AbirGallala, 2021-05-17:
In fact I can't find the right formula to write the command to move the robot via ros.
Do anyone know how I can write it with the new joints values to apply?

lyh458, 2021-06-16:
Introduction about how to command iiwa with iiwa_stack can be found [here](https://github.com/IFL-CAMP/iiwa_stack/wiki/commandrobot), two methods provided: 

- rostopic, e.g. command iiwa to **HOME** withe command line ``rostopic pub -1 /iiwa/command/JointPosition iiwa_msgs/JointPosition -- '{position: {a1: 0.0, a2: 0.0, a3: 0.0, a4: 0.0, a5: 0.0, a6: 0.0, a7: 0.0}}'``, or you can publish relative message to the topic in your cpp file.
- c++ api, example can be found in [iiwa_stack_examples](https://github.com/SalvoVirga/iiwa_stack_examples). BTW, if you have catkin_build error, try to use this [PR](https://github.com/SalvoVirga/iiwa_stack_examples/pull/11).

All work for me.

Mechazo11, 2022-10-28:
@lyh458 I am also facing this issue. I can read all the states and they are accurate. But publish command does nothing. I am seeing a warning message that position and torque sensors are not referenced. I master all the axes multiple times but these two warning messages do not go away. Could this be the cause?

lyh458, 2022-10-29:
@Mechazo11 I hasn't been used iiwa for a long time, have you try the following command or method? 

> Introduction about how to command iiwa with iiwa_stack can be found [here](https://github.com/IFL-CAMP/iiwa_stack/wiki/commandrobot), two methods provided:
>
> * rostopic, e.g. command iiwa to **HOME** [...]

---
> I master all the axes multiple times but these two warning messages do not go away. Could this be the cause?

I'm not sure if that's the reason. Maybe you can show more detailed about you warning.

Mechazo11, 2022-10-29:
@lyh458 I solved the issue of postition and torque sensor warnings by running the PositionandGSMReferencing application first. Then all the tiles on smartPad shows green. However, rostopic pub -1 like commands still does not actuate the robot. When I try to open the movit example I am getting an error saying a xarco file is not found. This may be a issue with how I had changed Robotnik's (creator of RBKAIROSba platform with a mounted a KUKA lbr iiwa 7) rosparameters. I am going to try on a separate ros melodic laptop to see if the "command" commands work and write back here