IFL-CAMP/iiwa_stack #270: Stuck while initializing Moveit! MoveGroupInterface

Oscar-B-Liang, 2021-07-28:
I am trying to use the MoveGroup interface in iiwa stack, but it seems the code hangs around while initailzing MoveGroupInterface. My source code is as simple as the following.

`#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>

#include <moveit_msgs/DisplayRobotState.h>
#include <moveit_msgs/DisplayTrajectory.h>

#include <moveit_msgs/AttachedCollisionObject.h>
#include <moveit_msgs/CollisionObject.h>
#include <ros/ros.h>

int main(int argc, char** argv)
{
    ros::init(argc, argv, "move_group_interface_tutorial");
    ros::NodeHandle node_handle;

    ros::AsyncSpinner spinner(1);
    spinner.start();

    static const std::string PLANNING_GROUP = "manipulator";

    ROS_INFO("Start initializing.");
    moveit::planning_interface::MoveGroupInterface move_group_interface(PLANNING_GROUP);
    ROS_INFO("Initialized.");

    moveit::planning_interface::PlanningSceneInterface planning_scene_interface;

    const moveit::core::JointModelGroup* joint_model_group = move_group_interface.getCurrentState()->getJointModelGroup(PLANNING_GROUP);
    return 0;
}`

and the launch file which starts this source code is

`<launch>
  <remap from="robot_description" to="iiwa/robot_description"/>
  <node name="moveit_test" pkg="moveit_test" type="moveit_test" output="screen"/>
</launch>`

I first run 
`roslaunch iiwa_stack moveit_planning_execution.launch`
then launched my own launch file. However, I got the following output:

`... logging to /home/liangby/.ros/log/53587aba-ef97-11eb-be7e-e39af1afea1d/roslaunch-liangby-17Z90P-G-AA76A3-46653.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://liangby-17Z90P-G-AA76A3:35157/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /trajectory_execution/controller_connection_timeout: 5.0

NODES
  /
    moveit_test (moveit_test/moveit_test)

ROS_MASTER_URI=http://localhost:11311

process[moveit_test-1]: started with pid [46669]
[ INFO] [1627471934.903575165]: Start initializing.
[ INFO] [1627471934.907294255]: Loading robot model 'iiwa7'...
[ INFO] [1627471934.907311573]: No root/virtual joint specified in SRDF. Assuming fixed joint`

Obviously, it is stuck at step in the cpp code

`moveit::planning_interface::MoveGroupInterface move_group_interface(PLANNING_GROUP);`

It cannot even execute its next line

`ROS_INFO("Initialized.");`

I tried on both ROS noetic and melodic, still having the same issue. I really wonder why does this happen. I can think of three possibilities.

1. I have to use ROS Kinetic to work on iiwa stack.
2. The iiwa_moveit package is not meant for me to program using python or cpp code, it is really just meant for GUI interface.
3. There is something wrong in my settings.

Can anyone please help me? This problem is driving me crazy.

exo-core, 2021-07-29:
> and the launch file which starts this source code is
> 
> `<launch> <remap from="robot_description" to="iiwa/robot_description"/> <node name="moveit_test" pkg="moveit_test" type="moveit_test" output="screen"/> </launch>`

Not sure if it solves the problem, but far as I know `remap` only works for topics, not for parameters.

a-mathieu, 2023-03-28:
Hi @Oscar-B-Liang, did you manage to fix this issue ? I'm stuck with the exact same problem for few days now. Thanks !