IFL-CAMP/iiwa_stack #272: There are no robot applications and background tasks available

civerachb-cpr, 2021-09-13:
I've just finished following the instructions from https://github.com/IFL-CAMP/iiwa_stack/wiki/sunrise_project_setup using the latest master branch and installed my project on the Kuka cabinet -- the installation process itself shows no errors nor warnings.

After restarting the cabinet, when I tap on the Applications menu on the tablet, I get the message "There are no robot applications and background tasks available."

[image]
_Apologies for the bad focus; with the glare my phone wouldn't focus on the tablet itself very well_

I'm pretty sure I've done something wrong, but I've gone through all of the steps a few times and I'm still getting the same problem in the end.

I've attached screenshots of my StationSetup.cat configuration & software tabs.  This is the first time I've set up a Kuka cabinet like this, so any insight would be greatly appreciated.

*Configuration*
I've edited the IP address since there will be 2 arms mounted to a robot (.50 and .51), and the ROS master is running on the 192.168.131.0/24 subnet
[image]

*Software*
[image]

*RoboticsAPI.data.xml*
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<RoboticsAPIData version="3">
    <world>
        <gravitation x="0.0" y="0.0" z="9.81"/>
    </world>
    <processDataContainer>
        <processData dataType="java.lang.String" defaultValue="right_iiwa" displayName="Robot Name" editableOnHmi="true" id="robot_name" value="right_iiwa"/>
        <processData dataType="java.lang.String" defaultValue="192.168.131.1" displayName="ROS Master IP" editableOnHmi="true" id="master_ip" value="192.168.131.1"/>
        <processData dataType="java.lang.String" defaultValue="11311" displayName="ROS Master Port" editableOnHmi="false" id="master_port" value="11311"/>
        <processData dataType="java.lang.String" defaultValue="192.168.131.50" displayName="Robot IP" editableOnHmi="false" id="robot_ip" value="192.168.131.50" visibleOnHmi="false"/>
        <processData dataType="java.lang.Boolean" defaultValue="false" displayName="Enable NTP" editableOnHmi="true" id="ntp" value="false"/>
        <processData dataType="java.lang.Boolean" defaultValue="false" displayName="Enable Debug Output" editableOnHmi="true" id="debug" value="false"/>
    </processDataContainer>
</RoboticsAPIData>
```

Like I said above, I'm pretty sure there's probably something I've missed, but having gone through the whole process 2-3 times now, I'm honestly not sure what it is.  If anyone can see anything obviously wrong with my configuration I'd really appreciate the assistance.  Thanks!

matthias-mayr, 2021-09-14:
I haven't used `iiwa_stack` in a while, but I think the only thing you are missing is the last step of the instructions:

> You can now install it (Station Setup -> Installation) and **then synchronize it.**

It's the button with the gray box and the blue arrows.

civerachb-cpr, 2021-09-14:
Thank-you!  That was it.  I feel appropriately stupid now.

closed 2021-09-14 by civerachb-cpr