IFL-CAMP/iiwa_stack #278: Too large error of trajectory tracking using moveIt

kirillin, 2022-01-31:
Hi guys!
I have a problem with controlling a robot with an environment with obstacles.
On fast trajectories, the robot deviates too much from the desired trajectory. In the moveIt, the trajectory is set correctly, but in real life the robot breaks the exactly same environment.

The scene we used is
[image]
[image]

I have a hypothesis that this is due to the default position control implementation in iiwa_hw package, instead of the position velocity control. If anyone has an intuition, how can I solve my problem? Thanks!

Kubasinska, 2022-12-07:
I think it's a duplicate of #284, Did you find a solution?

kirillin, 2023-07-11:
In fact, slow movements were enough for me at that time. And I didn't have enough time to dive deep into the problem.

In the following projects, I used the FRI interface without moveIt and everything worked great.

As for this problem, I think in iiwa_stack it need to send the trajectory from the moveIt to the controller and do retime using the internal interpolator of the robot.

But that's just my thoughts. I didn't solve the problem.

Also, I think there are another good solutions for iiwa that also can be test with moveIt
- https://github.com/RobotLocomotion/drake-iiwa-driver
- https://github.com/epfl-lasa/iiwa_ros
- https://github.com/ICube-Robotics/iiwa_ros2
- https://github.com/KCL-BMEIS/lbr_fri_ros2_stack