IFL-CAMP/iiwa_stack #279: Can we use moveit setup assistant to edit the file? unable to locate the URDF file in package. Expected File: /home/user/catkin_stackiiwa/src/iiwa_stack/iiwa_description/urdf/iiwa14.urdf

mf093087, 2022-02-03:
I want to use moveit setup assistant to edit the configuration file. However, it doesn't work for me. Can anybody help me?

unable to locate the URDF file in package. Expected File: /home/user/catkin_stackiiwa/src/iiwa_stack/iiwa_description/urdf/iiwa14.urdf

[image]

ymollard, 2022-02-03:
The URDF is generated by Xacro at runtime, but the setup assistant requires the URDF to be available.
You can generate it manually:
```bash
roscd iiwa_description/urdf
xacro iiwa14.urdf.xacro >iiwa14.urdf
```

Also make sure that the right IIWA version 14 or 7 is used there: https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_moveit/.setup_assistant#L4-L6

mf093087, 2022-02-04:
thanks, it works!