IFL-CAMP/iiwa_stack #280: RoboticsAPI.data.xml Parse Error

babbatem, 2022-02-09:
Hi all, 

We encountered a strange issue with the process data configuration file that I'd like to document. We were able to fix it after a bit of head-scratching. Here goes:

Upon starting ROSSmartServo, there were parse errors for src/RoboticsAPI.data.xml (after previously functioning properly). Process data was unable to load, frames weren't loading, mastering was unavailable, etc. We attempted to reinstall the project but installation failed with "reason: Unknown". 

Upon inspection, that file appears to have been corrupted during operation.  With mouse&monitor on the cabinet's Windows machine we edited the file to match the example provided. Then rebooting and installing the project with sunrise worked as usual. 

This happened simultaneously to two of our arms. Not sure what to make of it...have folks encountered similar issues before?

nevangeliouNYUAD, 2022-09-29:
I am having the exact same issue. Sunrise version 1.16

babbatem, 2022-09-29:
Glad to know we're not alone. We're on Sunrise 1.17.