IFL-CAMP/iiwa_stack #284: Using iiwa_moveit results in a jerky motion

daniellehot, 2022-03-21:
Hello. I configured the connection between the Sunrise workbench and the ROS machine according to the wiki and managed to connect with the robot without any issues. After connecting with the robot, I can echo the available topics just fine. However, if I try to move the robot using MoveIt, the robot moves in a jerky fashion. The MoveIt configuration files are unchanged, and MoveIt is launched using `roslaunch iiwa_moveit moveit_planning_execution.launch`. 

Interestingly, the jerkiness gets worse with lower velocity and acceleration scaling values. Furthermore, the jerkiness is more obvious for short trajectories. I tried multiple OMPL planners, but the result was the same. Currently, I am using RViZ to set goals for the robot. 

Has anyone else experienced this? If so, what can I do to fix this? I am using ROS Melodic, Ubuntu 18.04.

daeunSong, 2022-03-23:
Hi, 

I am having the same issue with ROS Melodic, Ubuntu 18.04, and Sunrise OS v1.11.
The result is the same whether I set goals through Rviz or commands (code).

I could not figure out the solution, yet. For a temporal solution, instead of using Moveit!, I decided to use messages/services of iiwa_stack. And this works fine with clean motions. 

After some inspection, I am expecting this might be related to the network issue. Could be also related to #282.
I am expecting that the robot is having a jerky motion as it fails to reach the real-time update rate of the command PC. 
I'd love to hear the opinions of others.

Thanks!

Daeun

Kubasinska, 2022-12-07:
I have the same problem.  Just a friendly ping, Have you found a solution? 

In my case the problem is present both in the JointPosition and JointPositionVelocity Hardware Interface.
This is what i get by registering messsages from /iiwa/status/JointPositionVelocity and /iiwa/command/JointPositionVelocity.

[image]

daniellehot, 2023-03-03:
It has been some time since I worked on this, and I apologize for not responding to the previous comments, but if I remember correctly, we ultimately resorted to using _iiwa\_stack_ services and actions to generate and execute robot trajectory. We used _move it_ to generate trajectory waypoints to prevent collisions with the environment. This method is not 100% reliable, as the trajectory between waypoints is not guaranteed to be collision-free (the robot's controller was not aware of the cell setup). However, such collisions were quite rare for our use and mostly the result of us enforcing a certain goal orientation of the end-effector. 

Let me know if you have more questions, but I plan to close this issue soon.

closed 2023-03-13 by daniellehot