IFL-CAMP/iiwa_stack #297: Diect publish joint positions

ChongTang09, 2023-03-30:
Hi, I just started learning how to use ROS to control KUKA iiwa 7 R800. I followed all the instructions and have successfully used ROSSmartservo app to connect my PC with iiwa robot arm via ROS. I can also subscribe position information from robot. However, when I tried to directly publish joint positions to iiwa, nothing happened. I checked nodes information and seems that iiwa has subscribed msgs but just not moved. What could be the possible reasons for this issue? Is there any additional driver or some settings needed if I want to do so?

ChongTang09, 2023-03-30:
Another question maybe should be the first question: do I have to have Sunrise Workbench installed on my PC side?

lyh458, 2023-03-30:
Right topic? Right namespace? Or others?

It would be convenient for us to track out your issue if you pass your code here as much as possible. 

---
> Another question maybe should be the first question: do I have to have Sunrise Workbench installed on my PC side?
- It's not necessary.